max-baz/wluma #150: [Feature request]: Manual base brightness slope instead of learning

mya4k, 2025-09-07:
### Please describe your feature request

Is it possible to set like values of als readings and correspond then to constant values. I.e. outdoors (100) = Brightness 50% and night (0) = 1%. And every value in between would be interpolated (at least linearly). The problem with learning is that if I suddenly choose that I want to use overall higher or lower brightness, I would need to retrain the algorithm, and by doing this all I would need to do is to just change those values.

It would also be useful to get the program store the associations of als readings and base brightness somewhere so that people could see if and how wluma has trained incorrectly and it's easier to adjust, would someone wish to use overall higher brightnesses

maximbaz, 2025-09-07:
Hello! 

The learned associations are stored in `~/.local/share/wluma/*.yaml` files, you can stop wluma, edit them and restart, and wluma will read your changes; and yes it's also possible to replace the learning algorithm with something else that lets you control the absolute brightness and only does the relative adjustments, see https://github.com/maximbaz/wluma#algorithm for details.

mya4k, 2025-09-07:
Hey, thanks for the reply, the Algorithm section does explain this but it only explains how to calibrate brightness reduction according to luma. I am not sure how to set lux instead. I guess I can just edit the prediction file instead tho (maybe info on this should be in the README)

mya4k, 2025-09-07:
Also can I replace threshold values like "outdoor" in that file with a numerical value or do I have to use them?

maximbaz, 2025-09-07:
The mapping between stringified values like "outdoor" and numerical values is defined in your config, e.g. https://github.com/maximbaz/wluma/blob/25d6574ca49f63b065f2674cd8b4e07e97d052e1/config.toml#L3

You can also make that more static, e.g. make lux value be dependent on the hour of the day, by using different [als] provider.

I think you can start by playing with the `yaml` files, changing them manually, perhaps making that file readonly or git committing it if you are afraid to irreversibly losing values, and see on which workflow you land, and we can try to make it easier.

Generally the idea behind the project is based on a premise that you don't "suddenly" change brightness without any apparent reason, if you are changing brightness it's because you are reacting on some external factor (currently most likely light around you or screen content), and as long as wluma also can see those factors, your "sudden change" will just make the trained model better, not worse.

mya4k, 2025-09-07:
Yes, I know about the threshold values. I was wondering if I could just use numerical values instead. But honestly, I could just add more threshold strings in the config, you're right.

The external factor at play is actually the fact that my screen is glossy, and sometimes I find the reflections more annoying (could be due to the bright clothes I am wearing today), that's the reason I want to play around with the training data, instead of having to retrain wluma. I think I just wanted more fine grained control and I did find your advice useful. I might add to the README about the prediction files later, if I have time. But thanks a lot for the help, closing this now.

closed 2025-09-07 by mya4k