max-baz/wluma #152: [Bug]: do just like MacOS does

yamato720, 2025-09-27:
### Steps for reproducing the issue

everytime i input "code ." in terminal, the folder always shows a "~" folder. even if i have remove it because it can't work on my ASUS Zenbook 14s on archlinux. it can't read the ALS while win11 can

### What is the buggy behavior?

this is the "~" folder 

[image]

### What is the expected behavior?

don't create it

### Logs

```shell
RUST_LOG=debug wluma
```

### Version

i use yay install it, maybe the latest until this mouth?

### Environment

```shell
LSB Version:    n/a
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a
Linux Archlinux 6.16.8-arch2-1 #1 SMP PREEMPT_DYNAMIC Sun, 21 Sep 2025 23:19:08 +0000 x86_64 GNU/Linux
egrep: warning: egrep is obsolescent; using grep -E
clang 20.1.8-1
lib32-vulkan-icd-loader 1.4.321.0-1
lib32-vulkan-intel 1:25.2.3-2
lib32-vulkan-mesa-device-select 1:25.2.3-2
rust 1:1.90.0-2
vulkan-icd-loader 1.4.321.0-1
vulkan-intel 1:25.2.3-2
vulkan-mesa-device-select 1:25.2.3-2
bash: dpkg: command not found
egrep: warning: egrep is obsolescent; using grep -E
WLR_DRM_NO_MODIFIERS=
```

maximbaz, 2025-09-27:
Hey, are you sure you created this issue in the correct repo...? 😅 I'm not really sure why you believe this is caused by wluma?

yamato720, 2025-09-27:
Maybe, but why this folder only created when input "code ." in terminal. Other programs that run throuth terminal input and get current path as its internal terminal path like vivado will not create the folder.

closed 2025-09-27 by yamato720